pmgducati/Blue-Retro-AIO-Units #3: Main Gerber file pin spacing

pceslayer, 2022-03-21:
Not sure this is because of the ESP3 boards I have on hand or because of how the PCB was printed, but my ESP32's pins do not line up with the PCB through holes. I have two different ESP32 models and they both have the same issue. I used OSH Park to print the PCB's and never had an issue with them before. 

I have a different version of the DevKitC on the way from DigiKey. Hopefully it'll fit.

[image]

pmgducati, 2022-03-21:
I used a NodeMCU-32 , the only difference is the lack of silkscreen pin callout on the top, which apparently makes it slightly thinner.  

I have purchased them from both places below and used them in multiple blue retro builds:

https://www.cytron.io/p-nodemcu-32s-wifi-bluetooth-ble-iot-dev-board

https://www.amazon.com/HiLetgo-ESP-WROOM-32-Development-Microcontroller-Integrated/dp/B0718T232Z/ref=sr_1_1?crid=PFTY2N59IPYA&keywords=HiLetgo+ESP-WROOM-32+ESP32&qid=1647893072&s=industrial&sprefix=hiletgo+esp-wroom-32+esp32%2Cindustrial%2C141&sr=1-1

pmgducati, 2022-03-21:
The BOM has been updated with the Amazon link,

closed 2022-03-23 by pmgducati

pceslayer, 2022-03-23:
Hi i'm attempting to edit the main board file in eagle to adjust the spacing and I'm missing the ESP32-DevKitC library.
Could you point me in the right direction? Thanks!

joeybaumgartner, 2022-09-25:
Just a note that I think the manufacturer changed the layout of the board linked in the BOM. I ordered one linked by the BOM and it does not fit the pinouts of the circuit board. I ended up ordering these and they work: https://www.amazon.ca/gp/product/B07QCP2451/.  

**NOTE**: These boards will not program unless you hold down the `IO0` button on the board itself when programming over USB. Updating firmware via the OTA interface appears to work just fine.