pmgducati/Blue-Retro-AIO-Units #14: Phillips CD-i PCB

pmgducati, 2022-04-19:
I do not have this system this PCB is compatible with, please put your findings here.

hipsofjw, 2022-06-14:
I'm a little confused about this PCB. What are the extra 4 holes for? They don't seem to fit a typical CD-i port. Also, I think you only need one cable since it has the same connector on each end.

pmgducati, 2022-06-14:
Hi, I am assuming you are referring to these holes:

[image]

That is for a USB mini connector, according to DarthCloud you need external power for the BlueRetro setup.  The sole purpose of that is to power the unit.

Here is the link to the comments:
https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-Cables-Build-Instructions#cd-i-adapter-cable

pmgducati, 2022-06-14:
I should add a callout on the PCB for that.  I will get it in the next revision.

pmgducati, 2022-06-14:
Missed if there too!  Thanks for catching it.  I will update both the PCB and BOM

hipsofjw, 2022-06-14:
One more thing while I've got you. There's one pad each for RTS and RXD for each port, but there are two RTS and RXD pins in each connector. Do I need to solder both to each pad?

pmgducati, 2022-06-14:
I don't believe so, it looks like all you need it P1 RTS/RXD from the front and P2 RTS/RXD from the Rear.  Check out the schematic from Darth's repo

https://github.com/darthcloud/BlueRetroHW/blob/master/DIY/CDI.pdf

pmgducati, 2022-06-14:
Also I updated the CD-i PCB but realized that the library I used had a obsolete part for the mini USB port.  I updated and the foot print did change.  When you buy one for your PCB with the old foot print just make sure to measure the distance from the front row of pins to the GND legs.  the pinout is standard and should fit whatever you find.  Im looking forward to hearing your about your findings on the PCB

hipsofjw, 2022-06-15:
Went sent the new gerber off to be printed. I'm in China, so it's only a couple of bucks. I'll let you know how it works once I set it up.

Thanks for quick responses and updates!

hipsofjw, 2022-06-15:
One additional thing: how are you supposed to plug in the micro USB while the PCB is in its case?

pmgducati, 2022-06-15:
Hey,
I completely missed that when modeling the housing.  I updated it by adding the cutout for power and offsetting the controller cable hole to the side.

I just pushed the commit that includes the changes.

[image]

hipsofjw, 2022-06-15:
That was fast! No need for a second wiring hole for P2?

pmgducati, 2022-06-15:
I mirrored the hole for P2:

[image]

Let me know what you think, you're the expert here ;)

pmgducati, 2022-06-15:
I pushed it to a branch here:
https://github.com/pmgducati/Blue-Retro-AIO-Units/tree/CD-i-housing-P2-Update

Here is the model if you want to look at it:
https://github.com/pmgducati/Blue-Retro-AIO-Units/blob/CD-i-housing-P2-Update/STL/Aux%20Bottom%20CD-i.STL

if it looks good I will merge it.

hipsofjw, 2022-06-15:
Looks good to me. Thanks for the update! I'll get one printed up soon.

pmgducati, 2022-06-15:
Merged!

pmgducati, 2022-09-28:
@hipsofjw did you finish this?  How is it working?

Nekimaliizgrada, 2022-11-19:
> I don't believe so, it looks like all you need it P1 RTS/RXD from the front and P2 RTS/RXD from the Rear.  Check out the schematic from Darth's repo
> 
> 
> 
> https://github.com/darthcloud/BlueRetroHW/blob/master/DIY/CDI.pdf
> 
> 

That's it

Nekimaliizgrada, 2022-11-19:
Phillip ate and drank what his brothers ate and drank. The disrespect of some people. Come on guys.

pmgducati, 2022-11-19:
@Nekimaliizgrada context?

Nekimaliizgrada, 2022-11-19:
Climax 😂😂😂😂Sent from my iPhone

closed 2022-12-20 by pmgducati