pmgducati/Blue-Retro-AIO-Units #15: PC-FX PCB

pmgducati, 2022-04-19:
I do not have this system this PCB is compatible with, please put your findings here.

hipsofjw, 2022-06-14:
I built one of these. Made the controller connectors with some housings I 3D-printed from thingaverse. Used a SNES cable and glued the connectors into the housing. Haven't tried two-player yet, but I did mess around with whatever fighting game I had and it works great!

hipsofjw, 2022-06-14:
I managed to pair both an Xbox Series X controller and a PS4 controller and it works! Nice!

This did lead me to a question though. To pair a second controller, one must enter inquiry mode with the first controller already paired. To enter inquiry mode, you press the BOOT button on the ESP32 itself. In order to do that, you need to take the main board out of its case. Is that right?

closed 2022-06-14 by pmgducati