pmgducati/Blue-Retro-AIO-Units #17: SNES PCB

pmgducati, 2022-04-19:
I do not have this system this PCB is compatible with, please put your findings here.

amaiorano, 2022-10-24:
So far this works perfectly for 2 players, with only 2 level shifters. I wanted it to work for 4 players, but the SNES extensions I got from AliExpress are missing 2 wires -- this is mentioned on [the last bullet point of the official instructions](https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-Cables-Build-Instructions#bill-of-materials-11).

Here's what it currently looks like:

[image]

I'm currently waiting on a lot of broken original SFC controllers that I can harvest the cables from so that they have the 7 wires all connected, and will test that multitap support works. I'll report my findings here.

pmgducati, 2022-10-25:
This is great to hear, looking forward to your updates.

RetroBobba, 2022-11-16:
I've made a full populated board with resistors for the PAL console and tried Micro Machines with 4 Players. It works fine :-)

Thanks for your work

pmgducati, 2022-11-16:
Thats Great to hear, thanks for the update!

closed 2022-11-16 by pmgducati