pmgducati/Blue-Retro-AIO-Units #18: Sega Dreamcast p

pmgducati, 2022-04-19:


joeybaumgartner, 2022-09-11:
I'm doing a build of this and hope to fully test this soon. Not sure if this is due to routing or not, but the P1 through P4 solder points are reversed compared to all of the other 4 player units I've assembled so far (N64 and GC). This just makes it so that the cables running out the 3D printed case cross over each other and harder to tighten up. I've just run the wires out straight and made a note to myself that P4 = P1, P3 = P2, P2 = P3, and P1 = P4.  Not sure if it's easer to update the PCB, or make an alternate box to 3D print.

pmgducati, 2022-09-12:
Hey,
Thanks for the heads up, I must have missed that when making the PCB.  I have made the changes and the pads now match the others and housing.
[image]

pmgducati, 2022-09-12:
Keep me updated on how your project goes!

corim123, 2022-09-24:
I'm working on some of the GC adapters(well waiting on DB25s) and plan on getting some of these boards next as I've got a Dreamcast kicking around somewhere.

joeybaumgartner, 2022-09-25:
Sorry for such a late reply on this. Still working with the original boards (ordered them before noticing the issue). I have had some success where I can connect a controller up to the Dreamcast, but cannot get it to recognize the Start button on either an XBox One controller or with a Wii-mote connected to a SNES Classic controller. 

@corim123 I have had no issues with the GC adapters so far, and I used the Wii-mote + SNES Classic controller for playing GBA games via GB Player and that's 1000 times better than using a GC controller.

pmgducati, 2022-09-28:
I just picked up a dreamcast and ordered the PCBs myself.  I will update when the PCBs arrive in a couple weeks

joeybaumgartner, 2022-09-28:
That sounds great! I'm interested to hear your results.

pmgducati, 2022-11-28:
@joeybaumgartner Finally got the PCBs assembled and my Dreamcast setup and I am not having any issues with my setup.  I have tested with 3 different 8bitdo controllers (SN30 Pro+, Pro 2 and N30), 2x PS4 controllers and 2x Xbox One controllers.  I played for about 4 hours with different combinations of all my controllers and single and multiplayer.  I tested Sonic adventure, zero gunner 2, MVC2, DOA 2, Mr Driller, Hydro Thunder and Daytona USA.

I am going to close this comment as I am not seeing any issues with so many games.  I you are still have issues please open a new issue.
[image]
[image]
[image]

closed 2022-11-28 by pmgducati