pmgducati/Blue-Retro-AIO-Units #42: Sega Genesis/Megadrive Multitap PCB

pmgducati, 2022-06-15:
I do not have this system this PCB is compatible with, please put your findings here.

DariensArcade, 2022-06-19:
Working:  Fully populated Multitap board working in 2 player mode in both 3/6 buttons (BR Setting Gamepad / Gamepad Alt.  respectively) on Genesis Model 1, 2, and 3.  

Further testing needed: I'm getting glitches attempting to use multi-tap modes but probably due to my controllers. i.e. Out of control scrolling with NHL '95 (EA Multi-tap. BR setting "alt")  

Will test further with other controllers/games/tap types

Reporting since no one else has.

pmgducati, 2022-07-01:
This is great to hear,  is NHL95 the only game you are seeing issues with?  Also is it on an original cart/flashcart?

closed 2022-09-12 by pmgducati