pmgducati/Blue-Retro-AIO-Units #45: Using Ps2 adaptor on an original Xbox

Harryunderhill, 2022-06-28:
Hi All

Ive made an PS/2 PSX adaptor which works great on both my PSX/PS2, how ever when I try 
To use it on my original Xbox with my PS/2 to Xbox adaptor, it powers up but nothing.

Ive been using the xbox to PS/2 adaptor on my Xbox for years and it works great, I just want
To use my wires less adaptor on it ?

I done the pin swap fix for the PS/2 adaptor, which works great on my ps/2 PSX.

any help would be great.

pmgducati, 2022-06-29:
It sounds like you are not getting enough power for the ESP to work correctly.   What Pad are you using for the PSX/PS2? (8V or 3V3?)  My Guess is that the xbox adapter wants to use the opposite that you are using and the LED is getting enough power to light up the the ESP is stuck in a boot loop due to the lack of consistent power.

Harryunderhill, 2022-06-29:
> It sounds like you are not getting enough power for the ESP to work correctly. What Pad are you using for the PSX/PS2? (8V or 3V3?) My Guess is that the xbox adapter wants to use the opposite that [...]

I’m using 3V3

pmgducati, 2022-06-29:
you could try to swap to the 8V pin to see if it works with the xbox adapter.

Harryunderhill, 2022-06-29:
Hi there I swapped the pins over to 8v and still nothing, I then tried both Xbox one and PS4 controller, don’t sync.
As soon as I plug the adaptor into the PS2 and turn both the PS4 and the Xbox one controller on they sync straight away.

any other ideas ?

pmgducati, 2022-07-01:
Is the adapter that you use still available?  I could pick one up and do some testing.  

For now, if you could remove the power connection (8v/3V3) from the controller side and use USB to power the ESP.  If it works with the adapter this will narrow it to a power delivery issue.

Harryunderhill, 2022-07-01:
> Is the adapter that you use still available? I could pick one up and do some testing.
> 
> For now, if you could remove the power connection (8v/3V3) from the controller side and use USB to power the ESP. If it works with the adapter this will narrow it to a power delivery issue.

Ive disconnected the power and powered the ESP externally and the controller syncs, however none of the buttons works or you can't navigate around the Xbox screen.

There is no brand on the adaptor either so don't no if you can still buy it.

xybots2022, 2022-07-10:
Hi I have exactly the same problem i bought a forcebox psx to xbox convertor and tried that with the ESP32 and I get exactly the same problem as Harry, ive tried powering the esp32 separately and the controller syncs but I cant navigate round the screen, any help would be great.

pmgducati, 2022-07-11:
@xybots2022 can you post a link to the adapter that you are using?  I will pick one up and do some testing.

xybots2022, 2022-07-25:
All I Do is search psx to xbox adaptors in eBay and I bought a couple from there which both work wired on the xbox but not with bluetooth.

thanks

pmgducati, 2022-11-14:
Without a specific unit to test against I will not be able to work on this issue

closed 2022-11-14 by pmgducati