pmgducati/Blue-Retro-AIO-Units #49: PIN Spacing Issue

joeybaumgartner, 2022-09-25:
Just a note that I think the manufacturer changed the layout of the board linked in the BOM. I ordered one linked by the BOM and it does not fit the pinouts of the circuit board. I ended up ordering these and they work: https://www.amazon.ca/gp/product/B07QCP2451/.

**NOTE**: These boards will not program unless you hold down the `IO0` button on the board itself when programming over USB. Updating firmware via the OTA interface appears to work just fine.

joeybaumgartner, 2022-11-14:
This is the link for the "corrected" version that would fit on the PCB. I did find this version did have an issue with flashing (I had to hold a button down on the unit), but otherwise this fit the pin-spacing. This is on Amazon Canada, rather than Amazon US, but the description for the one I got with wide spacing matched what was in the BOM. But I'm wondering if the seller had multiple versions.

pmgducati, 2022-11-29:
I eliminated the thinner version and went with the standard ESP32 footprint this should eliminate confusion in the future

closed 2022-11-29 by pmgducati