pmgducati/Blue-Retro-AIO-Units #51: DB25 male/female is inverse of official BlueRetro DIY instructions

amaiorano, 2022-09-29:
Hello,

I noticed that the choice of male or female DB25 connector is the inverse of what is specified on the official BlueRetro's site for DIY. For example, for the ESP32 module, [it says to use a female connector solder cup](https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-DIY-Build-Instructions#bill-of-materials), while your project uses a male one. Of course, the opposite is true for the cable adapters.

I know it's not a big deal, as one can simply swap the male/female DB25 connector as needed, but it might be worth mentioning somewhere in the README, especially for those of us (me) that have already built the official DIY ESP32 module, and would like to try interfacing it with a PCB-based one from this site.

Thank you for this awesome project!

P.S. Unrelated, but I think it would be nice if the README also included a screenshots showing the PCBs wired up _inside_ the 3D-printed shells, with the cover off, so that we can see how well it fits inside. Probably also worth mentioning that the cables would need to be threaded through the case before soldering to the PCB.

pmgducati, 2022-09-29:
This was an intentional design choice that I made. I would prefer to minimize pins that can be bent/damaged.  which is why I swapped the female/male.  I created a matching version of the ESP side with the male pins to go with all the systems.  The idea is that you would build the complete solution from here.  

This started as a personal project that I wanted to make for myself and throughout the development I had a few people ask me to post what I made/make additional systems.  

If you want to submit instructions, I will be happy to include them.

closed 2022-09-29 by pmgducati

amaiorano, 2022-09-29:
Makes sense, and thank you for sharing!