pmgducati/Blue-Retro-AIO-Units #60: NTSC NES Only A Button Working

hydrohs, 2022-12-14:
I just wired up the NES and SNES PCBs, but I’m having an issue with my NES version. I can connect controllers but the A button is the only input I get. I double checked continuity following the schematics and everything is connected correctly, I’m not sure what else to check.

hydrohs, 2022-12-14:
@pmgducati I’m tried with an NSO NES controller, NSO SNES controller, and a Dualshock 4

pmgducati, 2022-12-15:
PCB Looks good, Can you verify the pinout for the pigtails?
[image]

hydrohs, 2022-12-15:
That’s correct, yes. If it’s at all relevant my console is RGB modded, and I’m using an Everdrive.

hydrohs, 2022-12-20:
I think I found the issue, I'm almost positive one or both of my buffer chips were bad. I have no way to test them for sure, but I probably killed them with too much heat or something. I swapped them onto my SNES PCB (which was working) which then no longer worked. Ended up just assembling another from scratch and it's working perfectly. Your help and this project are much appreciated.

closed 2022-12-20 by hydrohs