pmgducati/Blue-Retro-AIO-Units #64: PCEngine 6 buttons

tawyy, 2023-03-24:
Hello,

I've made some of your PCB since it is exactly what I needed :) Thank you for everything !
I have tried Neo Geo and PCEngine so far. I need more cables for other systems, but i will soon. I could help you for issue 41, I have a RGB modded famicom AV which is my main famicom console for playing, and a NES PAL, my own NES from childhood....

So, I tried to play street fighter with your adapter and it works fine when playing with 2 buttons mode, but on 6 buttons mode > GameAlt, the controllers are not :( The controller is connected as 2nd player and can also control the first one where the bottom button seems to be always pressed in.  Am I miss something ? 

Thank you

pmgducati, 2023-03-24:
Hi,
Any testing that you can do with the Famicom and NES PAL would be AWESOME!  Put your findings in the issue.

As far as the 6 button issue goes,  I will do some testing this weekend to see if I can replicate the issue.  Just for reference, what controllers are you using?  Also can you post a picture of the PCB for PCE?

Thanks!

tawyy, 2023-03-24:
I found a solution to the 6 button problem... It seems weird at first, but I tried connecting the shield to ground and it solved all my problems! I read this on a HumbleBazooka thread somewhere... Not sure why it might affect 6 button mode, but it does!
I will try the PAL NES and my Japanese Famicom AV when I have the connectors.
Thank you.

pmgducati, 2023-03-24:
This is great to hear, I will test it as is (no shield to gnd) and the  your finding with my setup.  I will update the issue and make and needed changes for the PCB and call outs.

Thanks for digging in!

pmgducati, 2023-03-25:
I did some testing with and without the shield to GND and can verify this also.  I have added a callout on the silkscreen for the PCB.  It reads "Connect the Cable Shield to GND"

https://github.com/pmgducati/Blue-Retro-AIO-Units/pull/65

closed 2023-03-25 by pmgducati