pmgducati/Blue-Retro-AIO-Units #67: PSX/PS2 PCB has incorrectly numbered vias for L connector

amaiorano, 2023-04-30:
It looks like the numbering for the L connector vias on the PSX/PS2 PCB is flipped horizontally:

[image]

I was following the traces to make sure I connected my wires right, and was confused that the pad numbers didn't match up with the DB25 pin numbers as per darthcloud's schematic.

pmgducati, 2023-05-03:
The pin numbers are different, I swapped the male and female DB25 connectors so that the female version is on all the system adapters,  That way there is only male connector that could get bent pins and have to be repaired.  With the swap, the pin numbers are reversed on all the PCBs.  The traces all go to the correct pins on the main board and has been tested and verified to work as intended. 

Closing this comment.

closed 2023-05-03 by pmgducati

amaiorano, 2023-05-03:
> The pin numbers are different, I swapped the male and female DB25 connectors so that the female version is on all the system adapters, That way there is only male connector that could get bent pins [...]

Ah, I see, that makes sense. It's a bit unfortunate because the numbering of the pads on the silkscreen match up with what's in the official schematics, but not the DB25 pin numbers. Anyway, it does indeed work as expected. Thanks.