pmgducati/Blue-Retro-AIO-Units #69: No an issue with the hardware just a question

brendan1495, 2023-09-07:
Maybe a daft question but what does 

OD & PP stand for, looking at the neogeo , supergun, jamma adapter

Thanks

pmgducati, 2023-09-07:
Short answer is that OD is the one you need.  

Darthcloud has OD (Open Drain) and PP (Push Pull) for Neo Geo and the 2600/SMS versions.

Here is the explanation from his repo:

Typical parallel input system simply use a switch between input line and GND which pull it low on button press. The input line idle high via a pull-up resistor in the console. To best emulate this behavior use the open drain schematic.

If for a specific application you need the line to be actively be driven high you can follow the Push-Pull variant.

I made both to provide a complete collection of cables. I have the OD version for my Neo Geo and it works great.  I'm working on a internal version also.

brendan1495, 2023-09-07:
Thanks a lot for the explanation, I will get the OD version sent off for manufacture, look forward to the internal.

Again thanks

closed 2023-11-24 by pmgducati