pmgducati/Blue-Retro-AIO-Units #76: Open sch and brd

retrolover, 2024-03-17:
Hi, thanks by hard work you made.

I have some issues to open schematics or boards with Eagle, it trhows me : file not supported error.

Are urs designed made with eagle right?

pmgducati, 2024-03-17:
Yes, I used eagle to create the PCBs.  I have not run into a file unsupported error when opening my own or any others, sometimes GitHub adds a header to the files if you download them individually, try downloading the repo as a zip and see if that helps.

retrolover, 2024-03-17:
Thanks i Will try It.
For NeoGeo IS necessary level shifter?

Regars
Israel

pmgducati, 2024-03-17:
Yes, any parts listed in the BOM are required.  

One other note use the OD version.  The PP version is for non standard use cases.

retrolover, 2024-03-17:
Gotcha.
Then for neogeo i must use level shifter.

closed 2025-03-03 by pmgducati